WPIRoboticsEngineering/1001-romi-processor-shim #11: Clamp the analog signal coming back from the servo to 3.3v

madhephaestus, 2020-06-18:
We do not know what the voltage range is exactly, but we know we need to clamp it to 3.3v.

acamilo, 2020-06-18:
There might be a problem here. If this is the wiper of the pot then putting a zener on it could clamp it and prevent the voltage from going above 3.3v and keep the servo from making it's full range of motion.

If the servo is turned all the way over this can short the 5v rail through the pot causing damage. I recommend replacing this with a high value resistor like a 10k and letting the input pin clamp it to vcc.  I'm putting a schematic symbol for a zener but we can swap it out on the BOM with a resistor.

madhephaestus, 2020-06-18:
so a 10k in-line? that could work. Is there a pull-up on that pin on the board?

acamilo, 2020-06-18:
no, it goes right from a pin out to the servo's pot.

closed 2020-06-18 by acamilo