WPIRoboticsEngineering/1001-romi-processor-shim #14: Review pin-out and header placement of shim

acamilo, 2020-06-18:


madhephaestus, 2020-06-18:
Spread the user headers out a lot more to make room for SS, move the serial bus's off to the right into that free space

madhephaestus, 2020-06-18:
in fact, move all the headers to the space near the DevKitC's antenna, leaving lots of space for the devkitC heared labels

madhephaestus, 2020-06-18:
Pinout confirmed.

acamilo, 2020-06-18:
The USB comes in from the left and the antenna hangs over the side on the right.

acamilo, 2020-06-18:
i'm going to slide both modules over to the right and put the headers on the left. Sound good?

madhephaestus, 2020-06-18:
then they will interfere with the USB

acamilo, 2020-06-19:
The cables enter in through the right.

[image]

acamilo, 2020-06-19:
I think we have enough room for the labels as it is. I don't think moving the modules is a good idea.
I can move the headers a little.

acamilo, 2020-06-19:
Look at these labels. That's what i had in mind.
[image]

acamilo, 2020-06-19:
Also, The thing plus
[image]

acamilo, 2020-06-19:
Check out the board, I moved stuff around to free up some room.

madhephaestus, 2020-06-19:
looks good

acamilo, 2020-06-19:
Awesome

closed 2020-06-19 by acamilo