WPIRoboticsEngineering/1001-romi-processor-shim #16: Label convention for special function headers

madhephaestus, 2020-06-19:
Each pin should be labeled with its GPIO number, and its function. 

IE
```
VSPI_SCK/5
```

and

```
ECHO/14
```

Around typed headers, make a SS box with a label (like the 3001 motherboard had).

```
VSPI
HC-SR04
Line Sense L
Line Sense R
I2C
Servo
```

acamilo, 2020-06-19:
Can we entertain a more compact label configuration?

acamilo, 2020-06-19:
Perhaps, having the sensor headers labeled with the function (ultrasonic, spi, linefollowing)
Then, put the GPIO number (the one used in code) next to each pin.

And then, have a key or a note, to the side with a pin-out of the sensor.

acamilo, 2020-06-19:
One exception, the headers for a BUS (SPI,I2C) will be labed with the bus signal (SI,SO,SDA,SCL, ..etc)

acamilo, 2020-06-19:
Let me put together a draft for us to review.

acamilo, 2020-06-19:
Adafruit places the pin name next to the sensor.
[image]
[image]
[image]
[image]
[image]

acamilo, 2020-06-19:
What about something like this? This is for the ultrasonic.
* Polarity is indicated with - and+
* Trigger and Echo indicated with icons
* GPIO Pins Labeled
[image]

acamilo, 2020-06-20:
What about something like this?
[image]

acamilo, 2020-06-20:
I might want to pick a different font for the text. it is a little cramped.
Maybe we ditch the icons and just do GPIO numbers and labels?

acamilo, 2020-06-20:
For the other headers, How about something like this?
[image]

madhephaestus, 2020-06-20:
awesome!

madhephaestus, 2020-06-29:
There is no way for the user to see what pins are "availible". A SS label of how each pin is used, highlighting pins that are hard-routed and so not availible.

acamilo, 2020-07-10:
[image]

closed 2020-07-10 by acamilo