WPIRoboticsEngineering/1001-romi-processor-shim #37: Possibly move PWM pin

gcl8a, 2020-07-30:
If you swap the SERVO pin with M1_PWM, then it's marginally easier to control the PWM lines for both motors with the SAMD51. PA18 and PA19 default to the same TCC and TC, which would make timer (frequency) adjustment easier. You _can_ put PA17 and PA19 on the same TCC, but you have to do some magic to the Arduino libraries (e.g., edit variant.cpp for the board) or a little extra setup (hardly a big deal).

Not sure if swapping breaks anything on the ESP32, or even if it's worth it (there are plenty of TC/TCCs on the SAMD51), but if it's zero-loss elsewhere and only a #define change for the library, it might be worth it.

I haven't looked to see how the SAMD21 lines up. Happy to do that, if you want.

madhephaestus, 2020-08-10:
the esp has no hardware assigned pwm pins, so swapping is easy. the question i have is what eps pin is being pulled in here and where did it come from

madhephaestus, 2020-08-10:
could you rephrase this request as specific pin swaps you would like

madhephaestus, 2020-08-10:
we must deal with the fact that all pins are assigned. to move a pin, one must remove what was on it, and move it somewhere else. this is easy to do in the header, but just list off the specific swaps and i can test it.

gcl8a, 2020-08-10:
> Assigned #37 <https://github.com/WPIRoboticsEngineering/1001-romi-processor-shim/issues/37> to @gcl8a <https://github.com/gcl8a>.

Let’s not worry about it for now. If we end up using a SAMD chip, we’ll have plenty of time to adjust — C-term at the earliest. I’m swamped today and it’s more important to get this out than to spend time testing something esoteric.

closed 2020-08-10 by madhephaestus